washcroft/HttpGarageDoorController #1: Software for ESP8266 NodeMCU

Diederik98, 2017-09-12:
Hi, 

Thanks for your great plugin and arduino code! I copied your version and adjusted it to use it with an ESP8266. I also added OTA support. 

I'll post it as attachment.

Kind regards!

[OTA ESP8266 http garagedoor.zip](https://github.com/washcroft/HttpGarageDoorController/files/1297360/OTA.ESP8266.http.garagedoor.zip)

jeylites, 2017-09-22:
@Diederik98 

Thanks for the ESP8266 support. I tried to compile today but was stoped with an error. I seem to be missing a Tasker.h file. 

Build options changed, rebuilding all
/Users/Documents/Arduino/ESP8266_HTTP_OTA_GARAGEDOOR/ESP8266_HTTP_OTA_GARAGEDOOR.ino:8:20: fatal error: Tasker.h: No such file or directory
 #include <Tasker.h>
                    ^
compilation terminated.
exit status 1
Error compiling for board Generic ESP8266 Module.

Diederik98, 2017-09-22:
Glad you liked it!

You have to add the library first. 
You can find it on the main page under Dependencies.

Tasker, Bounce have to be installed. I'm not sure ArduinoJson is one of the standard libraries. Otherwise you also need to install that library.

jeylites, 2017-09-22:
I included the library through Arduino Manage library. It complies but I don't see any MQTT traffic on serial monitor. 

I see it has created a device in my router. The web page shows the following.

{ "result": 401, "success": false, "message": "The requested resource was unauthorised." }

Diederik98, 2017-09-22:
I always tested it with the homebridge accessory, because I don't know that much about http headers. Use same api key and check the homebridge log. Have you uncommented '#define DEBUG 1' in your code? Try to use a print statement when the garage is opened and toggle it with your accessory on your iOS device.

jeylites, 2017-09-22:
I uncommented #define DEBUG 1 and I was able to see the MQTT messages. It's working now. 

I just want to confirm the pins connection below. I connect a relay to light & door, the light works however not the door. 

What are the #define numbers corresponding to e.g.: #define D1 5 ?

DOOR RELAY OPEN/CLOSE = D2
SENSOR REED SWITCH OPEN = D8
SENSOR REED SWITCH CLOSE = D0
LIGHT = D4

Sketch "
#define D0 16
#define D1 5
#define D2 4
#define D3 0
#define D4 2
#define D5 14
#define D6 12
#define D7 13
#define D8 15
#define D9 3
#define D10 1
// LED ON ESP8266 on D4 or GPIO 2
const int LED_OUTPUT_PIN = D4;
const int DOOR_OUTPUT_OPEN_PIN = D2;
const int DOOR_OUTPUT_CLOSE_PIN = D2;
const int LIGHT_OUTPUT_PIN = D4;
const int LIGHT_INPUT_PIN = D1;
// not used, but I gave it a random pin, I did not want to change the code that much.
const int SENSOR_OPEN_INPUT_PIN = D8;
const int SENSOR_CLOSED_INPUT_PIN = D0;
"

Diederik98, 2017-09-26:
I'm sorry for my late answer. I hope it's working now! I used a NodeMCU ESP8266 and each pin had a label. I did that to make the wiring for my device easier.
I think the sensor is HIGH when opened, just print the value to be sure.

washcroft, 2017-10-04:
Thank you Diederik98. I have just released v2.0.0 of this project which includes support for ESP8266 boards (and more).

closed 2017-10-04 by washcroft