washcroft/HttpGarageDoorController #4: API Key header authorization FAILURE - header mismatch

jeylites, 2017-10-05:
I'm making progress washcroft... I currently stuck on the API key header authentication

What is the relation for the following 

HTTP_SERVER_APIKEY_HEADER[] = "MyOAuthConsumerKey";
HTTP_SERVER_OAUTH_CONSUMER_KEY[] = "MyOAuthConsumerKey";
HTTP_SERVER_OAUTH_CONSUMER_SECRET[] = "MyOAuthConsumerSecret";
  
AND

"httpHeaderName": "MyPassword",
"httpHeaderValue": "MyPassword",

WiFi Connection Lost - Connecting
Finding WiFi hardware......found!
Connecting to WiFi network...{"result":200,"success":true,"message":"OK","light-input":false,"light-requested":false,"light-state":false,"sensor-open":true,"sensor-closed":false,"door-state":"open"}
{"result":200,"success":true,"message":"OK","light-input":false,"light-requested":false,"light-state":false,"sensor-open":true,"sensor-closed":false,"door-state":"open"}
connected to HELLO
Obtaining IP address...obtained 192.xxx.xxx.xxx
WiFi Connection Complete

SSID: HELLO
Signal Strength (RSSI): -63 dBm
Client MAC: xc:cf:xf:fx:bx:ex
Station MAC: ax:bx:ex:ec:5x:ce
Client IP Address: 192.xxx.xxx.xxx
Gateway IP Address: 192.xxx.xxx.xxx
mDNS Hostname: garagedoorcontroller.local

Starting OTA updater / mDNS responderStarting HTTP web server...started!
Initalising clock...initalised at 2017-10-05 17:16:28 UTC
{"result":200,"success":true,"message":"OK","light-input":false,"light-requested":false,"light-state":false,"sensor-open":true,"sensor-closed":false,"door-state":"open"}
{"result":200,"success":true,"message":"OK","light-input":false,"light-requested":false,"light-state":false,"sensor-open":true,"sensor-closed":false,"door-state":"open"}

Accepted new client connection from 192.xxx.xxx.xxx:49178
GET /controller HTTP/1.1
MyOAuthConsumerKey: MyOAuthConsumerKey
host: garagedoorcontroller.local
Connection: close

API Key header authorisation FAILURE - header mismatch
HTTP/1.1 401 Unauthorized
Content-Type: application/json
Connection: close

**Secret.h**
const char WIFI_SSID[] = "SMILE";
const char WIFI_PASSWORD[] = "0162855518";
const char OTA_UPDATER_PASSWORD[] = "MyPassword";

const char HTTP_SERVER_APIKEY_HEADER[] = "MyOAuthConsumerKey";

const char HTTP_SERVER_OAUTH_CONSUMER_KEY[] = "MyOAuthConsumerKey";
const char HTTP_SERVER_OAUTH_CONSUMER_SECRET[] = "MyOAuthConsumerSecret";

**Homebridge Config.**

{
	"accessories": [{
		"accessory": "HttpGarageDoorController",
		"name": "Garage Door",
		"lightName": "Garage Light",

		"doorOperationSeconds": 0,

		"httpHost": "garagedoorcontroller.local",
		"httpPort": 80,
		"httpSsl": false,
		"httpStatusPollMilliseconds": 4000,
		"httpRequestTimeoutMilliseconds": 10000,

		"httpHeaderName": "MyPassword",
		"httpHeaderValue": "MyPassword",

		"oauthAuthentication": true,
		"oauthSignatureMethod": "HMAC-SHA256",
		"oauthConsumerKey": "MyOAuthConsumerKey",
		"oauthConsumerSecret": "MyOAuthConsumerSecret",
		"oauthToken": "MyOAuthToken",
		"oauthTokenSecret": "MyOAuthTokenSecret",

		"apiConfig": {
			"apiType": "HttpGarageDoorController"
		}
	}]
}

washcroft, 2017-10-05:
When using the HomeBridge module with this Arduino project, ```httpHeaderName``` should be "X-API-Key" - this isn't clear so I'll add that to the docs.

```oauthConsumerKey``` and ```oauthConsumerSecret``` should match in both HomeBridge module and Arduino project.

```oauthToken``` and ```oauthTokenSecret``` should be empty "" because they're not using in the Arduino project.

jeylites, 2017-10-06:
@washcroft 

After some tinkering, I manage to get it working. The APIKEY HEADER threw me off. I was thinking "MyPassword" was supposed to be the same password as OTA in the Arduino secret.h sketch but what it really meant was "MyAPIKey".

```
**Secret.h**
const char HTTP_SERVER_APIKEY_HEADER[] = "MyPassword";
**Config.json**
"httpHeaderValue": "MyAPIKey"

```
**RELAY hi/lo trigger** 
Is there a way to set the relay outputs to LOW-level trigger. The reason I'm asking is that the relays go from HI to LO after the Wemmos D1 mini comes out from a power cycle or reset which in turn opens the garage door.

**API**
What are the Json API and Generic / Plain API usage and difference?

**Timings**
`DOOR_SENSOR_REACT_TIME` & `DOOR_MAX_OPEN_CLOSE_TIME `

Should REACT time be the same or as MAX time or by half? 
React time do you mean as how fast the state changes from one or the other?

Thank you.

washcroft, 2017-10-08:
> Is there a way to set the relay outputs to LOW-level trigger. The reason I'm asking is that the relays go from HI - LO - HI after the Wemmos D1 mini comes out from a power cycle or reset which in turn opens the garage door.

The controller brings the relay pin ```HIGH``` for the ```DOOR_OUTPUT_PULSE_DELAY_TIME``` time and then puts it back ```LOW``` (to simulate a button press). How are you connecting it to your motor controller if LOW triggers it?

> What are the Json API and Generic / Plain API usage and difference?

The homebridge-http-garagedoorcontroller is designed to support other integrations, but if you're using this HttpGarageDoorController project, you don't need to use those since setting ```apiType``` to HttpGarageDoorController configures the API correctly.

> Should REACT time be the same or as MAX time or by half?

Some door open/closed sensors "bounce" or "flicker" when changing state (i.e. magnetic reed switches), the ```DOOR_SENSOR_REACT_TIME``` setting sets how long the sensors have to settle in their current state before using their values.

```DOOR_MAX_OPEN_CLOSE_TIME``` sets how long the motor usually takes to open or close the door. If the door hasn't completely opened or closed during this time (as indicated by the sensors), then the controller can report the door the "stopped" moving and is therefore left partially open.

> Lastly, On Homebridge to change the Icon from a garage to a switch, do I change the MDNS NAME & FRIENDLY to = lock or switch?

No, these fields just identify the devices on the network. Currently, the module will always report the unit to HomeKit as a "Garage Door service". This may change with a feature update soon.

jeylites, 2017-10-08:
Is there a way to set the relay outputs to LOW-level trigger. The reason I'm asking is that the relays go from HI - LO - HI after the Wemmos D1 mini comes out from a power cycle or reset which in turn opens the garage door.

> The controller brings the relay pin HIGH for the DOOR_OUTPUT_PULSE_DELAY_TIME time and then puts it back LOW (to simulate a button press). How are you connecting it to your motor controller if LOW triggers it? 

Currently, I have my hardware setup the same way as yours except I'm using the Wemmos D1. 

Everything works fine - except the relay triggers everytime the unit  restarts - causing my garage door to open (relay is connected to garage door push button {open/close} )

The relay is in a 'high' state during normal operation, so when the power goes out, it triggered the relay. 

[image]

The kind of relay I'm using [RELAY](http://www.ebay.co.uk/itm/5V-1-2-4-8-Channel-Relay-Board-Module-for-Arduino-Raspberry-Pi-ARM-AVR-DSP-PIC-/252051910091?var=&epid=1642428054&hash=item3aaf76edcb:m:mdbiogK3bG3MfWb_sjDpLvw

washcroft, 2017-10-08:
I use a similar relay board.

As mentioned above, in this project the relay output is kept ```LOW``` and is only ```HIGH``` for a brief moment. Do you have the relay wired to your motor controller correctly (try swapping the NO and NC connection).

jeylites, 2017-10-08:
Currently, the push button contact is connected to relay NC pins since the
relay is always energized when connected to ESP8266 during operation.

washcroft, 2017-10-08:
The push button input of your door motor should be connected to the relay common and NO. NO means open circuit when the relay is not energised, and we want to close the circuit (push the button) when the relay is energised.

The relay should not be energized when idle (or at boot), when the MCU is powered off, when disconnected etc... Therefore, this is a relay board issue since it's giving you the opposite behaviour by energising on LOW input (i.e. default = ON).

I noticed a jumper on the relay board you linked to, does that change the behaviour to energise on HIGH input (i.e. default = OFF)?

jeylites, 2017-10-08:
I tried the jumper, it’s nothing more than an enable or disable relay
feature.

I also tried a new D1 and Relay board, the behavior still exists.

I’m wondering if would it fix the issue if get a relay with  hi/lo option.

washcroft, 2017-10-08:
> I’m wondering if would it fix the issue if get a relay with  hi/lo option.

I updated my reply on the GitHub issue. However, this is what you need to do, use a different type of relay board which you can trigger (energise) the relay with 3.3v HIGH.

washcroft, 2017-10-08:
> I’m wondering if would it fix the issue if get a relay with  hi/lo option.

I updated my reply on the GitHub issue. However, this is what you need to do, use a different type of relay board which you can trigger (energise) the relay with 3.3v HIGH.

Previous reply:

> It looks like you'll need a different relay board, or some circuitry to invert the signal. Whilst this could be changed in the code (so that it's permanently HIGH and goes LOW to trigger), this wastes energy in the relay and could present a safety/security issue (i.e. the door might unexpectedly open on MCU crash, power supply issue).
> 
> Here's what I used (but these are everywhere, just look for 3.3v high level trigger):
> 
> http://www.ebay.co.uk/itm/3-3V-5V-10A-Relay-Module-for-Arduino-Nano-Due-Raspberry-Pi-ESP8266-Optocoupler-/272272819712

jeylites, 2017-10-08:
I have made some developments. Because the relay module operates on 5V and ESP8266 operate on 3.3V, I used a 2N2222 NPN transistor to shift levels,  like the circuit below. So now the relay only energize when Open or Close is pushed on homekit App. 

[image]

You mentioned there is a way to permanently invert the trigger in the code, what changes should I make?

```
    digitalWrite(pin, HIGH);
    delay(DOOR_OUTPUT_PULSE_TIME);
    digitalWrite(pin, LOW);
```

The other problem is to prevent the relay to trigger upon ESP8266 startup. This translates to the gate opening or closing. I was looking into a delay circuitry which is tied to trigger relay VCC. Its little overkill but it works. 

just wondering, Is there a way to do this in the code?

jeylites, 2017-10-08:
Another update on relay triggering upon startup. Turns out, it's a common problem with these ESP8266. Some guy in blog suggested to put a 10k on the gpio pin (D1)  then other end of resistor to ground for pulldown. 

Everything work as it should be now. I'm still going to get a 3.3 relay as well. Also I was looking at the SSR, which does 3V to 32VDC  [Solid State Relays ](http://www.ebay.com/itm/SSR-25-DA-25A-DC-AC-24V-380V-Solid-State-Relay-for-PID-Temperature-Controller-/191989321890?hash=item2cb37490a2:g:mE4AAOSwWxNYrWEq)

washcroft, 2017-10-08:
1) Getting a 3.3v HIGH triggered relay would be my preferred option; or

2) Alter the circuit as you suggested, although I suspect it'd be easier to alter the relay board, simply by swapping around the input connections to the optoisolator); or

3) Alter the code:

https://github.com/washcroft/HttpGarageDoorController/blob/master/GarageDoorController.cpp#L33

becomes:

```
digitalWrite(pin, LOW);
delay(DOOR_OUTPUT_PULSE_TIME);
digitalWrite(pin, HIGH);
```

You might also need to add this line here, so that the trigger is HIGH at boot:

https://github.com/washcroft/HttpGarageDoorController/blob/master/GarageDoorController.cpp#L59

```
digitalWrite(DOOR_OUTPUT_OPEN_PIN, HIGH);
```

jeylites, 2017-10-08:
The changes you suggested worked. I no longer need the pulldown and the relay is triggering right. However, I'm going to purchase the 3.3V type High trigger just to keep it safe. 

Thank you Warren!

washcroft, 2017-10-08:
No problem :)

closed 2017-10-08 by washcroft

rjblake, 2018-03-04:
I have the same/similar 2-way relay and the same problem. I've amended code for the door relay and it seems to work ok, but the light relay always gets pulled low and turned on at boot/restart. I guess I'll have to try get hold of another module, but perhaps an enhancement request would be to allow the options as a configuration

jeylites, 2018-03-04:
@rjblake 

you should! The last thing you want is to have the garage door fail on you. I got the 3ch as linked below it has worked great ever since. Currently using stock code with the only alteration on output pins to work on Wemos D1 mini.

[1CH 3.3V Relay](https://www.ebay.com/itm/1PCS-3V-Relay-High-Level-Driver-Module-optocouple-Relay-Module-for-Arduino-NEW/381733076159?hash=item58e1107cbf:g:8asAAOSwgZ1XrEw7)

[3CH 3.3V Relay](https://www.ebay.com/itm/3-Channel-Relay-Module-With-Optocoupler-Isolation-Compatible-3-3V-5V-Signal/311751431759?hash=item4895d5564f:g:fx8AAOSwj85YPyEN)

washcroft, 2018-03-04:
@rjblake It's not possible to change what happens on the outputs whilst the MCU boots, of course we can control the outputs once it has booted into user code.

For this reason, better to get a module that works the right way around.

rjblake, 2018-03-05:
@washcroft then I’m really confused as previously have the ESP booting with ESPEasy firmware and it boots without any change to relay state; I.e. it remains in off state