washcroft/HttpGarageDoorController #5: Auto closing

davidkinders, 2017-10-31:
Hello,

I have a problem: When the door is open (sensor triggerd) the door automatically is closing around 42 seconds. Can't find the problem. 

Kind regards
David

washcroft, 2017-10-31:
Hello David,
Do you have DOOR_AUTO_CLOSE_TIME set in your config? It should be 0 if you don't want to use that feature.

davidkinders, 2017-10-31:
The time was set to 5min (30000), so i did't think this was the problem. But that seems the problem, thanks. 

BTW great code!

Kind regards
David

washcroft, 2017-11-01:
I wonder if your sensors aren't being read properly? The door should only close after DOOR_AUTO_CLOSE_TIME has elapsed  since the door state last become "open" (open sensor active, closed sensor inactive).

closed 2017-12-16 by washcroft