washcroft/HttpGarageDoorController #6: don't know how to start the project

FilD201, 2017-12-06:
hi, I don't understand what I specifically need to do to make this work, can you explain which code needs to go where?

washcroft, 2017-12-16:
This is a standard Arduino project, download the whole project, open HttpGarageDoorController.ino in the IDE, configure as per documentation and then upload to your chosen hardware.

closed 2017-12-16 by washcroft

sparklingsparklers, 2018-02-18:
@washcroft There is really a lack of said documentation...

washcroft, 2018-02-18:
@sparklingsparklers There is ample documentation in README.md, this is **project specific** documentation, not an introduction to micro controller programming and development.

sparklingsparklers, 2018-02-18:
@washcroft 

After downloading the project, creating the secret.h file, and downloading the dependencies, this is the error im getting. 

Arduino: 1.8.5 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 115200"

Build options changed, rebuilding all
Archiving built core (caching) in: /var/folders/9q/7t76cjg50vj1hzjh_ygqd6l00000gn/T/arduino_cache_337983/core/core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,UploadSpeed_115200_c3854604d461a8e339a634f719585bd3.a
sketch/HttpGarageDoorController.ino.cpp.o:(.text._Z41__static_initialization_and_destruction_0ii$part$0+0x18): undefined reference to `HttpWebServer::~HttpWebServer()'
sketch/HttpGarageDoorController.ino.cpp.o:(.text._Z41__static_initialization_and_destruction_0ii$part$0+0x1c): undefined reference to `WiFiHelper::~WiFiHelper()'
sketch/HttpGarageDoorController.ino.cpp.o: In function `__static_initialization_and_destruction_0':
/var/folders/9q/7t76cjg50vj1hzjh_ygqd6l00000gn/T/arduino_modified_sketch_877919/HttpGarageDoorController.ino:61: undefined reference to `HttpWebServer::~HttpWebServer()'
sketch/HttpGarageDoorController.ino.cpp.o: In function `~WiFiServer':
/Users/alan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/libraries/ESP8266WiFi/src/WiFiServer.h:50: undefined reference to `WiFiHelper::~WiFiHelper()'
sketch/HttpGarageDoorController.ino.cpp.o:(.text.setup+0x10): undefined reference to `WiFiHelper::enable_led(unsigned char, unsigned char, unsigned char, bool)'
sketch/HttpGarageDoorController.ino.cpp.o:(.text.setup+0x14): undefined reference to `HttpWebServer::enable_apikey_header_auth(char const*)'
sketch/HttpGarageDoorController.ino.cpp.o:(.text.setup+0x18): undefined reference to `HttpWebServer::enable_oauth_auth(UDP&, char const*, char const*, unsigned short, unsigned short, unsigned short)'
sketch/HttpGarageDoorController.ino.cpp.o: In function `setup':
/var/folders/9q/7t76cjg50vj1hzjh_ygqd6l00000gn/T/arduino_modified_sketch_877919/HttpGarageDoorController.ino:76: undefined reference to `WiFiHelper::enable_led(unsigned char, unsigned char, unsigned char, bool)'
/var/folders/9q/7t76cjg50vj1hzjh_ygqd6l00000gn/T/arduino_modified_sketch_877919/HttpGarageDoorController.ino:66: undefined reference to `HttpWebServer::enable_apikey_header_auth(char const*)'
/var/folders/9q/7t76cjg50vj1hzjh_ygqd6l00000gn/T/arduino_modified_sketch_877919/HttpGarageDoorController.ino:83: undefined reference to `HttpWebServer::enable_oauth_auth(UDP&, char const*, char const*, unsigned short, unsigned short, unsigned short)'
/var/folders/9q/7t76cjg50vj1hzjh_ygqd6l00000gn/T/arduino_modified_sketch_877919/HttpGarageDoorController.ino:88: undefined reference to `WiFiHelper::is_connected()'
/var/folders/9q/7t76cjg50vj1hzjh_ygqd6l00000gn/T/arduino_modified_sketch_877919/HttpGarageDoorController.ino:88: undefined reference to `WiFiHelper::connect()'
sketch/HttpGarageDoorController.ino.cpp.o:(.text._Z11connectWiFiv+0x14): undefined reference to `HttpWebServer::begin()'
sketch/HttpGarageDoorController.ino.cpp.o:(.text._Z11connectWiFiv+0x18): undefined reference to `Clock::update(bool)'
sketch/HttpGarageDoorController.ino.cpp.o:(.text._Z11connectWiFiv+0x1c): undefined reference to `Clock::now_utc()'
sketch/HttpGarageDoorController.ino.cpp.o:(.text._Z11connectWiFiv+0x20): undefined reference to `Clock::get_formatted_time(long, char*, unsigned int)'
sketch/HttpGarageDoorController.ino.cpp.o:(.text._Z11connectWiFiv+0x30): undefined reference to `WiFiHelper::is_connected()'
sketch/HttpGarageDoorController.ino.cpp.o: In function `connectWiFi()':
/var/folders/9q/7t76cjg50vj1hzjh_ygqd6l00000gn/T/arduino_modified_sketch_877919/HttpGarageDoorController.ino:96: undefined reference to `WiFiHelper::connect()'
/var/folders/9q/7t76cjg50vj1hzjh_ygqd6l00000gn/T/arduino_modified_sketch_877919/HttpGarageDoorController.ino:108: undefined reference to `HttpWebServer::begin()'
/var/folders/9q/7t76cjg50vj1hzjh_ygqd6l00000gn/T/arduino_modified_sketch_877919/HttpGarageDoorController.ino:109: undefined reference to `Clock::update(bool)'
/var/folders/9q/7t76cjg50vj1hzjh_ygqd6l00000gn/T/arduino_modified_sketch_877919/HttpGarageDoorController.ino:110: undefined reference to `Clock::now_utc()'
/var/folders/9q/7t76cjg50vj1hzjh_ygqd6l00000gn/T/arduino_modified_sketch_877919/HttpGarageDoorController.ino:123: undefined reference to `Clock::get_formatted_time(long, char*, unsigned int)'
sketch/HttpGarageDoorController.ino.cpp.o:(.text._Z17getJsonDeviceInfoPcj+0x30): undefined reference to `WiFiHelper::get_mac(char*, unsigned int)'
sketch/HttpGarageDoorController.ino.cpp.o:(.text._Z17getJsonDeviceInfoPcj+0x34): undefined reference to `WiFiHelper::get_client_ip(char*, unsigned int)'
sketch/HttpGarageDoorController.ino.cpp.o: In function `List':
/Users/alan/Documents/Arduino/libraries/ArduinoJson-master/src/ArduinoJson/Data/List.hpp:30: undefined reference to `WiFiHelper::get_mac(char*, unsigned int)'
/Users/alan/Documents/Arduino/libraries/ArduinoJson-master/src/ArduinoJson/Data/List.hpp:30: undefined reference to `WiFiHelper::get_client_ip(char*, unsigned int)'
sketch/HttpGarageDoorController.ino.cpp.o: In function `StaticStringBuilder':
/Users/alan/Documents/Arduino/libraries/ArduinoJson-master/src/ArduinoJson/Deserialization/../Serialization/StaticStringBuilder.hpp:14: undefined reference to `HttpWebServer::send_response(Client&, unsigned short, unsigned char c [... cut ...]
/Users/alan/Documents/Arduino/libraries/HttpGarageDoorController-master/GarageDoorController.cpp:182: undefined reference to `striendswith(char const*, char const*)'
/Users/alan/Documents/Arduino/libraries/HttpGarageDoorController-master/GarageDoorController.cpp:187: undefined reference to `striendswith(char const*, char const*)'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.