washcroft/HttpGarageDoorController #9: Compile errors

rjblake, 2018-02-12:
Just downloaded the zip and tried to compile for an ESP8266. Arduino IDE came back with a whole stack of errors. Guess I've got something wrong somewhere

Arduino: 1.8.5 (Mac OS X), Board: "Adafruit HUZZAH ESP8266, 80 MHz, 4M (1M SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 115200"

In file included from sketch/GarageDoorController.h:16:0,
                 from sketch/GarageDoorController.cpp:25:
sketch/src/Utilities/Utilities.h:11:18: warning: extra tokens at end of #ifdef directive [enabled by default]
 #ifdef LOG_LEVEL && LOG_LEVEL >= 1

                  ^
In file included from sketch/src/HttpWebServer/../../src/Sha/sha256.h:4:0,
                 from sketch/src/HttpWebServer/HttpWebServer.h:18,
                 from sketch/GarageDoorController.h:17,
                 from sketch/GarageDoorController.cpp:25:
sketch/src/HttpWebServer/../../src/Sha/sha256_config.h:36:0: warning: "HASH_LENGTH" redefined [enabled by default]
 #define HASH_LENGTH 32
 ^
In file included from sketch/src/HttpWebServer/../../src/Sha/sha1.h:4:0,
                 from sketch/src/HttpWebServer/HttpWebServer.h:17,
                 from sketch/GarageDoorController.h:17,
                 from sketch/GarageDoorController.cpp:25:
sketch/src/HttpWebServer/../../src/Sha/sha1_config.h:38:0: note: this is the location of the previous definition
 #define HASH_LENGTH 20
 ^
sketch/GarageDoorController.cpp:322:18: warning: extra tokens at end of #ifdef directive [enabled by default]
 #ifdef LOG_LEVEL && LOG_LEVEL >= 3

                  ^
In file included from sketch/GarageDoorController.h:13:0,
                 from sketch/GarageDoorController.cpp:25:
sketch/config.h:11:39: warning: large integer implicitly truncated to unsigned type [-Woverflow]
 const uint16_t DOOR_AUTO_CLOSE_TIME = 300000;

                                       ^
sketch/config.h:31:50: warning: large integer implicitly truncated to unsigned type [-Woverflow]
 const uint16_t OAUTH_TIMESTAMP_VALIDITY_WINDOW = 300000;

                                                  ^
In file included from sketch/GarageDoorController.h:16:0,
                 from /Users/richard/Documents/Arduino/HttpGarageDoorController/HttpGarageDoorController.ino:46:
sketch/src/Utilities/Utilities.h:11:18: warning: extra tokens at end of #ifdef directive [enabled by default]
 #ifdef LOG_LEVEL && LOG_LEVEL >= 1

                  ^
In file included from sketch/src/HttpWebServer/../../src/Sha/sha256.h:4:0,
                 from sketch/src/HttpWebServer/HttpWebServer.h:18,
                 from sketch/GarageDoorController.h:17,
                 from /Users/richard/Documents/Arduino/HttpGarageDoorController/HttpGarageDoorController.ino:46:
sketch/src/HttpWebServer/../../src/Sha/sha256_config.h:36:0: warning: "HASH_LENGTH" redefined [enabled by default]
 #define HASH_LENGTH 32
 ^
In file included from sketch/src/HttpWebServer/../../src/Sha/sha1.h:4:0,
                 from sketch/src/HttpWebServer/HttpWebServer.h:17,
                 from sketch/GarageDoorController.h:17,
                 from /Users/richard/Documents/Arduino/HttpGarageDoorController/HttpGarageDoorController.ino:46:
sketch/src/HttpWebServer/../../src/Sha/sha1_config.h:38:0: note: this is the location of the previous definition
 #define HASH_LENGTH 20
 ^
In file included from /Users/richard/Documents/Arduino/HttpGarageDoorController/HttpGarageDoorController.ino:43:0:
sketch/config.h:11:39: warning: large integer implicitly truncated to unsigned type [-Woverflow]
 const uint16_t DOOR_AUTO_CLOSE_TIME = 300000;

                                       ^
sketch/config.h:31:50: warning: large integer implicitly truncated to unsigned type [-Woverflow]
 const uint16_t OAUTH_TIMESTAMP_VALIDITY_WINDOW = 300000;

                                                  ^
/Users/richard/Documents/Arduino/HttpGarageDoorController/HttpGarageDoorController.ino: In function 'void loop()':
/Users/richard/Documents/Arduino/HttpGarageDoorController/HttpGarageDoorController.ino:205:19: warning: converting to non-pointer type 'int' from NULL [-Wconversion-null]
     if (client != NULL) {
[... 139 lines cut ...]

                                                                                                                                      ^
sketch/src/HttpWebServer/HttpWebServer.cpp:521:131: warning: passing NULL to non-pointer argument 4 of 'size_t percent_encode(const char*, size_t, char*, size_t)' [-Wconversion-null]
         size_t oauth_consumer_secret_encoded_size = percent_encode(oauth_consumer_secret, oauth_consumer_secret_length, NULL, NULL);

                                                                                                                                   ^
sketch/src/HttpWebServer/HttpWebServer.cpp:531:122: warning: passing NULL to non-pointer argument 4 of 'size_t percent_encode(const char*, size_t, char*, size_t)' [-Wconversion-null]
         size_t oauth_token_secret_encoded_size = percent_encode(oauth_token_secret, oauth_token_secret_length, NULL, NULL);

                                                                                                                          ^
sketch/src/HttpWebServer/HttpWebServer.cpp: In static member function 'static void HttpWebServer::send_response(Client&, uint16_t, const uint8_t*, size_t)':
sketch/src/HttpWebServer/HttpWebServer.cpp:695:23: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
   char *head_format = "HTTP/1.1 %d %s\r\nContent-Type: application/json\r\nConnection: close\r\n\r\n";

                       ^
sketch/src/HttpWebServer/HttpWebServer.cpp:702:23: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
   char *body_format = "{\"result\":%d,\"success\":%s,\"message\":\"%s\"}";

                       ^
exit status 1
Error compiling for board Adafruit HUZZAH ESP8266.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

sparklingsparklers, 2018-02-21:
Ever get to the bottom of this?

washcroft, 2018-02-27:
@rjblake @sparklingsparklers fixed in the latest commit, issue caused by changes in ESP8266 release v2.4

commit 6fe003bc (2018-02-27): Fixes #9
closed 2018-02-27 by washcroft