washcroft/HttpGarageDoorController #10: Enhancement Request: Add temperature sensor support (e.g. DHT22, BME280, etc.)

rjblake, 2018-03-09:
Still waiting for me eBay relay delivery and was thinking it might be nice to also be able to have the temperature of the garage reported. Haven't managed to wrap my head around how to include in the webserver and also to the Homebridge setup.

rjblake, 2018-03-27:
Problem sorted with the relay (triggered high version). I've added an SHT31 (temperature & humidity sensor) and adjusted the code to work with this. I considered a DHT22, but the reaction time is slow, so thought it better to go with a more 'reactive' sensor. However, I need to look at your accompanying Homebridge code to include reading of the sensor; as only in test mode still and have not included the OAuth authentication (i.e. only reads info when not secure). Once I've got this sorted, I'll create a new fork and submit pull request if you want to include.

rjblake, 2018-04-17:
Have had little time to look into this, but have pondered an alternative approach. As I'm using Domoticz for other purposes and it does a good job of pulling various different HA aspects together, it may make more sense in my case to look into using MQTT to at least publish information on door-state, light-state and temp/humidity to Domoticz so I have visibility there too (Homekit/Homebridge does not provide cover for all the devices/type of devices I have) and have an option of a more comprehensive approach. Now need to see where I need to add relevant MQTT pub/sub code without breaking it

washcroft, 2018-05-23:
Adding support for a temperature sensor is a good idea, to complement the light "addon". This and MQTT status publishing is on my todo list.