washcroft/HttpGarageDoorController #13: MQTT Publishing

jeylites, 2019-05-21:
@washcroft 

I can't be grateful enough, it's been 3 years now since I have been using your project for my home automation. I was wondering if the following updates are possible. For the most part, I have Homekit running all my home automation,  Home Assistant is used for tedious rules, Amazon Alexa Echo for voice commands. 

- Is there a way to bridge the gap between all these technologies? For example, garage door controller sending MQTT messages to Home Assistant or Alexa

Another fun addon will be to have a pulsing buzzer warning people the gate is about to open or a least Alexa voicing out through its speakers.

washcroft, 2019-05-21:
I did plan to add MQTT support to this project, but unfortunately I'm struggling to find the time - I will get around to it, at some point. Although I'm not keen on the idea of being able to control the door from MQTT alone due to the security risk, but it definitely makes sense for status updates.

I also like your idea of the alarm/warning which would be relatively easy to add.

jeylites, 2019-05-27:
Appreciate it. Hope something could be done soon.