washcroft/HttpGarageDoorController #15: Compiling Issues Arduino Library

jeylites, 2020-04-15:
@washcroft 

I hope all is well during these uncharted times.  I needed to change my Wemos D1 earlier today, tried to compile only to run into some errors. 

It seems that there's an incompatibility issue with ArduinoJson library. So I downgraded to ver.5 which has to lead me to another problem with WifiHelper

**ArduinoJson Issue**

```
HttpGarageDoorController:146:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6
   StaticJsonBuffer<200> jsonBuffer;
           ^
exit status 1
StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

```

**Wifihelper Issue**

```

sketch/src/WiFiHelper/WiFiHelper.cpp: In member function 'bool WiFiHelper::connect()':
sketch/src/WiFiHelper/WiFiHelper.cpp:202:23: error: ambiguous overload for 'operator!=' (operand types are 'wl_status_t' and 'const IPAddress')
     if (WiFi.status() != INADDR_NONE) {

                       ^
sketch/src/WiFiHelper/WiFiHelper.cpp:202:23: note: candidates are:
sketch/src/WiFiHelper/WiFiHelper.cpp:202:23: note: operator!=(int, uint32_t {aka unsigned int}) <built-in>
sketch/src/WiFiHelper/WiFiHelper.cpp:202:23: note: operator!=(int, u32_t {aka long unsigned int}) <built-in>
sketch/src/WiFiHelper/WiFiHelper.cpp:202:23: note: operator!=(int, int) <built-in>
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.
```

jeylites, 2020-04-17:
@washcroft 

I was able to recompile based on the following version but it would be nice to have an updated version. 

WifiHelper issue was due to the ESP Board Version. Only works on 2.4.2, anything above would give an error.

**HttpGarageDoorController**
HTTP Controller Firmware Version 2.0.0

**Arduino Library** 
ArduinoJson  Version 5.1.0
Arduino OTA Version 1.0.0
Wifi101 Version 0.11.2
Wifi 101 OTA Version  1.0.0

**Board Manager**
ESP8266 Version  2.4.2

**Arduino Compiler** 
Version 1.8.12 MacOsx

jeylites, 2020-05-07:
@washcroft 

Hello, just checking in....

jeylites, 2021-09-14:
Hello, Any thoughts on updating this lovely project? I still can't get it to work with my garage door. :(

lagunacomputer, 2025-12-27:
Nevermind, I got it working again.